madmartin/Jarolift_MQTT #1: "Set and generate Serial" checkbox has no effect

sidddy, 2018-06-13:
Hello,

I'm trying to change the serial of my jarolift_mqtt setup, but the checkbox "Set and generate Serial" on the system.html page does not seem to have any effect: Even after changing the serial and enabling the checkbox, the shutters still react on commands without re-learning.

Any idea?

Thanks!!

madmartin, 2018-06-13:
Hi,
there is a known issue with this button on the web UI with (at least) firefox - he does send the right things to the webserver.  (When you try to edit the channel names in the "shutter" dialog - same thing)
You can do two things:
1) wait a short time to the next release
2) try Chrome or Chromium- that works - but with another caveat: The new serial numbers are applied to the channel (what you cannot see in the WebUI) but the new start serial number is not stored and is *seems* that nothing has happened. This is a fault I found yesterday.

I propose Option 2), but PLEASE write down you new chosen serial number. 

And a short feedback would be nice.

Bye!

sidddy, 2018-06-13:
Hi!

Thanks for your fast feedback!!

I was using chromium for my tests, but still I didn't see any effect when enabling the checkbox.

I checked in the developer tools: the browser correctly sends the enabled checkbox, so frontend seems to be fine. I briefly checked the corresponding coding in html_api.h, but could not really find any reference to set_and_generate_serial (except for the "get config" part). Are you sure the coding in html_api is complete wrt set_and_generate_serial?

BR,
Sidddy

madmartin, 2018-06-13:
I think you are falling into a "work in progress" hole. That's exactly something I am currently working on.
Let me clean up my code and commit the remaining work in ~30mins.

sidddy, 2018-06-13:
hehe, sure, thanks for your support!! :-)

madmartin, 2018-06-13:
next commits are pushed, give it a try!

sidddy, 2018-06-13:
Woohoo, it worked!! All shutters work perfectly now with my newly set serial. Thanks again for your help!!

closed 2018-06-13 by sidddy