madmartin/Jarolift_MQTT #15: FirmwareUpdate

sisamiwe, 2018-08-18:
Hi,
is there a chance to be able to update the firmware via OTA?
Maybe you can add a the firmware update into the system menu?
THX

madmartin, 2018-08-20:
Hello @sisamiwe 
of couse OTA is technically possible, though nobody did write code for it until now.
Your example https://github.com/sidoh/esp8266_milight_hub/tree/1.8.1 has no OTA code (at least I cannot find it). Please point to exact filename and line numbers.

sisamiwe, 2018-08-21:
Hi,
sorry, but I'm not so familiar with code. 
I made a screen shot, where the firmware update is located:
[image]
Would this help?

meier81, 2018-08-21:
Hi, i think the problem with the update via the web interface is that you have to precompile all the different boards, i would find the update on the arduino-ide or platformio better

madmartin, 2018-08-21:
@sisamiwe yes, that helps a little. I was missing the typical OTA libraries from the ESP core.  @meier81 is right with his remark - we would need different binaries for a bunch of typical devices, but - what I think makes much more work - I need to integrate the SPIFF data into the binary image. This is a big step necessary before OTA (or "update over WLAN") is possible. Also the layout of the configuration memory must be changed completely. 
So it is a lot to do before I see this to implement.

sisamiwe, 2018-08-21:
@madmartin 
The advantage of an OTA update is, that you don't need to connect the dongle to the PC. It can stay where it is. Also for the other challenges the milight hub might be a good expample to learn from.
Michael