madmartin/Jarolift_MQTT #19: Jarolift_MQTT_v0.7 - not conneted to config ESP Page

HomeAutoUser, 2019-01-11:
Hallo,
vielen Dank für dein Projekt und ich ich wollte das ganze heute mal umsetzen.
Auf meinen NodeMC habe ich deine Jarolift_MQTT-0.7 geflasht und auch das Data Verzeichnis "hochladen" lassen. Das Device geht in den AccessPointMode und ich sehe ihn auch. 

Beim Login des Pointes komme ich zwar drauf aber keine Page funktioniert weil keine Ports offen sind.
Wo liegt mein Fehler oder ist noch ein Bug existent?

MfG

madmartin, 2019-03-27:
Wenn der ESP im AccessPoint-Mode ist, dann bietet er einen DHCP Server an (der gibt Deinem Endgerät eine IP aus dem Bereich 192.168.4.x) und einen Webserver auf http://192.168.4.1
Nach das WLAN verbunden ist, mußt Du einen Web-Browser starten und die Webseite http://192.168.4.1 ansteuern.
Mit etwas Glück - wenn Du zum Konfigurieren ein neueres Android-Telefon oder Tablet verwendest, dann führt Dich die Erkennung von "WiFi Anmeldeseite" von Android Dich direkt auf die http://192.168.4.1

Heimatkanal, 2019-05-15:
Zuerst danke für Deine Arbeit!
Zuerst ein paar Sachen:
1. Arduino IDE meckert wenn  das Verzeichnis  Arduino/**Jarolift_MQTT-master** heißt (so wird es entpackt). Will die Jarolift_MQTT.ino drin nicht annehmen sondern in Jarolift_MQTT Verzeichnis haben.
Nach umbenennen von dem Verzeichnis Jarolift_MQTT-master ind Jarolift_MQTT wird die .ino akzeptiert.
2. Deine verlinkte SPIFFS Datei funktioniert nicht mehr in neuesten Arduino IDE. Es kommt Fehler "Not found". Diese hier funktioniert:
[https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.4.0](url)

Jetzt was mache ich aber falsch? Alles so gemacht wie Du schreibst. Am Ende aber blinkt nichts, habe nur einen Wifi Point "ESP-XXXXXX", unverschlüsselt und kann die http://192.168.4.1 nicht öffnen wenn ich mich mit dem Point verbinde. Bekomme IP 192.168.4.2 also DHCP funktioniert.
Hardware:
Wemos D1 mini
Einstellungen:
Board: "LOLIN(WeMos) mini D1 R2 & mini"
Flash Size: "4M (1M SPIFFS)"
lwIP variant: "v2 Lower Memory"
CPU Frequency: "80 MHz"
Upload speed: "115200"

Flashen endet ohne Fehler.
Danke
Heimatkanal

madmartin, 2019-05-15:
Moin,
Ehrlich gesagt hab ich den Eindruck Du hast einiges durcheinandergeworfen.

1. Arduino IDE meckert wenn das Verzeichnis Arduino/Jarolift_MQTT-master heißt  - keine Ahnung - wenn das bei Dir so ist - kann ich nichts dazu sagen?!?
2. die verlinkte SPIFFS Datei. - gibt ne neue Version - verstehe. Werde ich beizeiten in der Doku anpassen.

Meine generellen Anmerkungen:
* Versuche bitte wenn es nicht klappt mit der Arduino IDE 1.8.5 zu arbeiten. Meine Doku bezieht sich darauf, ich habe selbst die neuere Version noch nicht benutzt und kann für nichts garantieren.
* Bitte öffne einen "Seriellen Monitor" und schau dort was passiert. Der Sketch schreibt einiges auf seiner seriellen Konsole. 
* Bei Fehlverhalten vom ESP: einemal komplett löschen, das heißt den gesamten Flash mit Nullen vollschreiben. Die Arduino IDE kann das nicht (wenigstens nicht die 1.8.5), ich habe unter Linux immer das "esptool.py" genutzt. Das steht - meine ich - aber auch alles in der Doku.

Das wichtigste ist: was passiert auf dem seriellen monitor / Konsole !!!!

Heimatkanal, 2019-05-15:
Hi,
Arduino 1.8.5 meckert auch wegen Jarolift_MQTT**-master** Verzeichnis. Ist aber kein Problem mit umbenennen.
Ich habe mit:
[http://iot-bits.com/erase-esp32-flash-completely/](url)
Erase gemacht
Danach Arduino 1.8.5 als Administrator in Win 10 gestartet, Wemos D1 mini an USB2 angeschlossen (vorher USB3) und noch mal nach Deinen Punkten geflasht. Nach dem Sketch flashen (konsole):
Bold = war in rot, MAC habe ich hier geändert

**Archiving built core (caching) in: C:\Users\my\AppData\Local\Temp\arduino_cache_212410\core\core_esp8266_esp8266_d1_mini_xtal_80,vt_flash,exception_disabled,ssl_all,eesz_4M1M,ip_lm2f,dbg_Disabled,lvl_None____,wipe_none,baud_115200_64c3fc0d568bac6c4b1c187f8a606584.a**
Der Sketch verwendet 362496 Bytes (34%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 37404 Bytes (45%) des dynamischen Speichers, 44516 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
**esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM15
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 3c:71:bf:2c:XX:XX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 366656 bytes to 259020...

Writing at 0x00000000... (6 %)
Writing at 0x00004000... (12 %)
Writing at 0x00008000... (18 %)
Writing at 0x0000c000... (25 %)
Writing at 0x00010000... (31 %)
Writing at 0x00014000... (37 %)
Writing at 0x00018000... (43 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (56 %)
Writing at 0x00024000... (62 %)
Writing at 0x00028000... (68 %)
Writing at 0x0002c000... (75 %)
Writing at 0x00030000... (81 %)
Writing at 0x00034000... (87 %)
Writing at 0x00038000... (93 %)
Writing at 0x0003c000... (100 %)
Wrote 366656 bytes (259020 compressed) at 0x00000000 in 23.3 seconds (effective 125.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...**

Seriell Monitor hat gezeigt:
1970-01-01 09:00:00 CEST [INFO] - starting Jarolift Dongle v0.7
1970-01-01 09:00:00 CEST [INFO] - ESP-ID 2926380 // ESP-Core  2_5_1 // SDK Version 2.2.1(cfd48f3)
1970-01-01 09:00:00 CEST [INFO] - read config from EEPROM . . .  no configuration found in EEPROM!!!!
1970-01-01 09:00:00 CEST [INFO] - write config to EEPROM
1970-01-01 09:00:01 CEST [INFO] - default config applied
1970-01-01 09:00:01 CEST [INFO] - initializing the CC1101 Transceiver. If you get stuck here, it is probably not connected.

Danach Wemos USB Kabel raus und rein = reset, und SPIFFS geflasht. Konsole:

[SPIFFS] data    : C:\Users\my\Downloads\arduino-1.8.5-windows\arduino-1.8.5\Jarolift_MQTT\data
[SPIFFS] size    : 1004
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/favicon.png

/index.html

/log.html

/scripts.js

/shutter.html

/style.css

/system.html

[SPIFFS] upload  : C:\Users\my\AppData\Local\Temp\arduino_build_747572/Jarolift_MQTT.spiffs.bin
[SPIFFS] address  : 0x300000
[SPIFFS] reset    : nodemcu
[SPIFFS] port     : COM15
[SPIFFS] speed    : 115200
[SPIFFS] python   : C:\Users\my\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1\python.exe
[SPIFFS] uploader : C:\Users\my\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\tools\upload.py

esptool.py v2.6

Serial port COM15

Connecting........_____.

Chip is ESP8266EX

Features: WiFi

MAC: 3c:71:bf:2c:XX:XX

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 4MB

Compressed 1028096 bytes to 33923...

Writing at 0x00300000... (33 %)
Writing at 0x00304000... (66 %)
Writing at 0x00308000... (100 %)
Wrote 1028096 bytes (33923 compressed) at 0x00300000 in 5.2 seconds (effective 1589.4 kbit/s)...

Hash of data verified.

Leaving...

Hard resetting via RTS pin...

Danach wieder USB Kabel raus und rein.
Am Ende wieder nur ESP_XXXXXX Wifi Point (Unverschlüsselt) mit 192.168.4.1 Gateway aber weiterhin WebUi nicht erreichbar.

madmartin, 2019-05-16:
Moin,

danke für das mitschneiden und senden des Konsol-Output/Serieller Monitor .... aber... genau da wo es interessant wird hört es auf.
Eine Frage: Hat Dein ESP Modul keinen Reset-Button?  Bitte einmal den Seriellen Monitor aufmachen und dann den Reset-Button drücken (wenn man immer den USB Stecker raus- und rein-steckt geht immer die COMx Schnittstelle weg und muß sie neu einstellen und den Monitor wieder öffnen - dann verpasst man immer den Anfang vom Start des ESP. Zumindest ist das bei mir unter Linux so.)

Die Ausgabe

1970-01-01 09:00:00 CEST [INFO] - starting Jarolift Dongle v0.7
1970-01-01 09:00:00 CEST [INFO] - ESP-ID 2926380 // ESP-Core 2_5_1 // SDK Version 2.2.1(cfd48f3)
1970-01-01 09:00:00 CEST [INFO] - read config from EEPROM . . . no configuration found in EEPROM!!!!
1970-01-01 09:00:00 CEST [INFO] - write config to EEPROM
1970-01-01 09:00:01 CEST [INFO] - default config applied
1970-01-01 09:00:01 CEST [INFO] - initializing the CC1101 Transceiver. If you get stuck here, it is probably not connected.

Ist doch schon mal ganz vielversprechend.
Und der letzte Satz ist übrigens kein Scherz - "initializing the CC1101 Transceiver. If you get stuck here, it is probably not connected." 
Das bedeutet: wenn es hier nicht weitergeht ist der CC1101 Transceiver wahrscheinlich nicht (richtig) angeschlossen und die Initialisierung bleibt einfach stecken.

Also - bitte nochmal den seriellen MOnitor aufmachen - was kommt nach "initializing the CC1101 Transceiver" ?

Grüsse
Martin

Heimatkanal, 2019-06-08:
Hallo,
Könnte erst jetzt alles richtig verbinden. Ich sehe jetzt AP "Jarolift Dongle" ist da, kann die Config Seite benutzen aber es speichert meine WLANP SSID und Password nicht. Es bleibt bei MYSSID und Admin Zugang über Jarolift Dongle. Hier serieller Monitor:

SDK:2.2.1(cfd48f3)/Core:2.5.1=20501000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
1970-01-01 09:00:00 CEST [INFO] - starting Jarolift Dongle v0.7
1970-01-01 09:00:00 CEST [INFO] - ESP-ID 2926380 // ESP-Core  2_5_1 // SDK Version 2.2.1(cfd48f3)
1970-01-01 09:00:00 CEST [INFO] - read config from EEPROM . . . config version 2 found
1970-01-01 09:00:00 CEST [INFO] - initializing the CC1101 Transceiver. If you get stuck here, it is probably not connected.
1970-01-01 09:00:00 CEST [WARN] - Admin-Mode enabled!
1970-01-01 09:00:00 CEST [WARN] - starting soft-AP ... scandone
del if0
usl
mode : softAP(xx:xx:bf:2c:a7:xx)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
 Ready
1970-01-01 09:00:00 CEST [WARN] - Access Point <Jarolift-Dongle> activated. WPA password is 12345678
1970-01-01 09:00:00 CEST [WARN] - you have 180 seconds time to connect and configure!
1970-01-01 09:00:00 CEST [WARN] - configuration webserver is http://192.168.4.1
1970-01-01 09:00:02 CEST [INFO] - HTTP server started
add 1
aid 1
station: d8:9a:34:92:c5:ff join, AID = 1
 File not found: did you upload the data directory?
1970-01-01 09:01:35 CEST [WARN] - Admin-Mode disabled, soft-AP terminate ...station: d8:9a:34:92:c5:ff leave, AID = 1
rm 1
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
usl
mode : null
 success
1970-01-01 09:01:35 CEST [INFO] - WiFi connecting to MYSSID
mode : sta(xx:xx:bf:2c:a7:xx)
add if0
scandone
no MYSSID found, reconnect after 1s
1970-01-01 09:01:38 CEST [INFO] - WiFi station disconnected
reconnect

SPIFFS vorher geladen:
[SPIFFS] data    : C:\Users\my\Downloads\arduino-1.8.5-windows\arduino-1.8.5\Jarolift_MQTT\data
[SPIFFS] size    : 1004
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/favicon.png

/index.html

/log.html

/scripts.js

/shutter.html

/style.css

/system.html

[SPIFFS] upload  : C:\Users\my\AppData\Local\Temp\arduino_build_278994/Jarolift_MQTT.spiffs.bin
[SPIFFS] address  : 0x300000
[SPIFFS] reset    : nodemcu
[SPIFFS] port     : COM14
[SPIFFS] speed    : 115200
[SPIFFS] python   : C:\Users\my\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1\python.exe
[SPIFFS] uploader : C:\Users\my\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\tools\upload.py

esptool.py v2.6

Serial port COM14

Connecting....

Chip is ESP8266EX

Features: WiFi

MAC: xx:xx:bf:2c:a7:xx

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 4MB

Compressed 1028096 bytes to 33923...

Writing at 0x00300000... (33 %)
Writing at 0x00304000... (66 %)
Writing at 0x00308000... (100 %)
Wrote 1028096 bytes (33923 compressed) at 0x00300000 in 5.2 seconds (effective 1585.4 kbit/s)...

Hash of data verified.

Leaving...

Hard resetting via RTS pin...

Heimatkanal, 2019-06-10:
Ich habe es. Post von Delphin0810 war wichtig. Ohne die Änderung hat WEBUI NICHTS gespeichert, sondern sofort neu gestartet. Änderung (name egal) in system.html:

<input type='button' class='siimple-btn siimple-btn--navy' value="save" onclick="writeConfig('save')" **name="save"**>
<input type='button' class='siimple-btn siimple-btn--grey' value="restart the dongle" onclick="writeConfig('restart')" **name="boot"**>

Hab aber trotzdem Probleme mit dem einlernen von TDRR01W.
Hat Jemand es damit Erfolg gehabt?

madmartin, 2019-06-12:
> Ich habe es. Post von Delphin0810 war wichtig. 

Wo ist dieser Post?

> Ohne die Änderung hat WEBUI NICHTS gespeichert, sondern sofort neu gestartet. Änderung (name egal) in system.html:
> 

Welcher Web-Browser wurde benutzt? Name des Browsers, Version bitte?

> <input type='button' class='siimple-btn siimple-btn--navy' value="save" onclick="writeConfig('save')" **name="save"**>
> <input type='button' class='siimple-btn siimple-btn--grey' value="restart the [...]

Steht doch im Wiki....   "known to work... " [https://github.com/madmartin/Jarolift_MQTT/wiki/Devices](url)

Wenn es nicht funktioniert... liegt es daran daß der ESP+CC1101 dergestalt senden daß der TDRR01W das nicht versteht. 
Woran es genau liegt gilt es herauszufinden:

- die richtigen Keys eingetragen? Ohne die geht's überhaupt nicht.
- Was passiert im seriellen Log wenn Tasten von den vorhandenen Fernbedienungen gedrückt werden? Sinnvolle sachen sind z.b. wenn mehrmals dieselbe Taste gedrückt wird auch im seriel log immer dieselbe Seriennummer als Absender auftaucht.  Wirre Daten lassen auf falsche Keys oder kaputte/schlechte Antenne schließen.
- wenn der Empfang schlecht ist, kann die Antenne vom CC1101 schlecht sein. Wie sieht die Platine vom CC1101 aus? So wie bei den Bildern im Wiki bei den "known to work devices"?
- ist die Platine vom CC1101 eine 400MhZ Variante oder 800MHz? die 800MHz Varianten werden sehr wahrscheinlich nicht funktionieren, steht aber auch irgendwo in den ganzen Anleitungen.

TheNoltens, 2019-08-14:
Ich habe das gleiche Problem. Wenn ich auf den Button save klicke, startet der ESP neu und alle Einstellungen sind nicht gespeichert worden. Habe es mit Chrome auf dem Handy ausprobiert.

krisha, 2020-01-20:
Bei mir kam auch immer ein WDT Reset beim Speichern. Mit dem Fix von delphin0850 funktioniert es: #22

kizur, 2021-05-08:
Hallo madmartin,

Habe probleme mit anlernen den MQTT version. Ich habe schon 2 module betrieb genommen, gegenseitig getestet, wenn im webui eine button drücke kommt auf der andere im seriel fenster das gesendete command folge bei "up" code 0x8, "down" 0x2, "stop" 0x4. Mit original FB auch gleiche art kommen die tasten kode, aber ist egel welche master_msb _lb kod stelle, die button kode kommen immer durch. Anlernen kann leider nicht!!! Habe doch falsche Maser_msb _lsb? kannst bitte mir helfen?