madmartin/Jarolift_MQTT #20: No Wifi connection

corch1988, 2019-03-01:
Hi! Thanks for this awesome project.

After some tinkering I think, I got the MSB and LSB Codes right. So I wanted to test it. But it seems like the ESP8266 does not connect to my wifi or the one of my father.
My SSID contains a space (" ") and the one of my father a dot ("."). Could this lead to this error?

For testing purpose I left the MQTT settings empty. Maybe that?

corch1988, 2019-03-01:
It looks like my settings are not saved. When I enter MYSSID and the password and I click on save, after a refresh of the page, I have to enter SSID and password again.

corch1988, 2019-03-02:
So using the correct ESP8266 Library Version (2.4.1) it works now.

It is mentioned in the Readme. Read first, ask later.

closed 2019-03-02 by corch1988

madmartin, 2019-03-04:
Hello @corch1988,
I am reading this today and want to say thanks for closing the issue and telling us your solution.
Martin