madmartin/Jarolift_MQTT #25: Gruppensteuerung

sisamiwe, 2019-07-28:
Hi,
would it be possible to also integrate group control?
As per the original [post ](http://www.bastelbudenbuben.de/2017/04/25/protokollanalyse-von-jarolift-tdef-motoren/) it should be possible?

It would be very beneficial to have the group control to be able to start all shutters within a room at the same time.

What do you think?

AIXruth, 2022-09-28:
Group control is possible in the following way:
- Use a remote controller like the TDRC16
- Select all shutters as group control
- Set all shutters into learn_mode (press a+b, then 8xstop) (all shutters will give feedback)
- Press the learn button for a desired channel (e.g. I used channel 0 for group control) (all shutters will give feedback)
Now all shutters are controlled by the selected channel.

dewenni, 2022-10-09:
@AIXruth thanks for that hint.
This is simple and it works fine!