madmartin/Jarolift_MQTT #42: Can not fnd the device in the network

atze09, 2021-07-01:
Hello i installed it on my wemod d1 and successfully got into the webinterface then changed to my wifi settings.
After this i searched for it in my ip range but could not find it.

The serial monitor gives me the following output:
1970-01-01 02:00:00 CEST [INFO] - starting Jarolift Dongle v0.7
1970-01-01 02:00:00 CEST [INFO] - ESP-ID 13683746 // ESP-Core  3.0.1 // SDK Version 2.2.2-dev(38a443e)
1970-01-01 02:00:00 CEST [INFO] - read config from EEPROM . . . config version 2 found
1970-01-01 02:00:00 CEST [INFO] - initializing the CC1101 Transceiver. If you get stuck here, it is probably not connected.
1970-01-01 02:00:00 CEST [INFO] - WiFi connecting to DS17
1970-01-01 02:00:00 CEST [INFO] - HTTP server started
1970-01-01 02:00:07 CEST [INFO] - WiFi station disconnected

Maybe you can help me out.
Can i see the actual ip address.

atze09, 2021-07-01:
I tested the hotspot on my mobile and was able to connect to it. Must be an issue with my network settings.
Tried to change the settings a few times, but could not get it working. Can you help me?

AIXruth, 2022-09-25:
Your issue might be related to #47. Have you checked this?