madmartin/Jarolift_MQTT #43: Smooth way to add shade command to home assistant?

atze09, 2021-07-01:
What's the best way to implement the shade command into home assistant config?

Is there a smoother way than this config (mqtt publish)
- type: button
    tap_action:
      action: call-service
      service: mqtt.publish
      service_data:
        payload: SHADE
        topic: cmd/jarolift/shutter/5
      target: {}
    entity: cover.buro

stalakerob, 2021-07-01:
I'm not using SHADE but since its a cover you better configure it as mqtt cover. See here [https://www.home-assistant.io/integrations/cover.mqtt/](url)

That's how I did it:

`cover:
  - platform: mqtt
    name: "Jarolift"
    command_topic: "cmd/jarolift/shutter/0"
    availability_topic: "tele/jarolift/LWT"
    state_topic: "stat/jarolift/shutter/0"
    value_template: >
      {% if value == "100" %}
        open
      {% else %}
        closed
      {% endif %}    
    payload_open: "UP"
    payload_close: "DOWN"
    payload_stop: "STOP"
    payload_available: "Online"
    payload_not_available: "Offline"`

yasars, 2021-07-04:
I have instead of payload_stop: "STOP", payload_stop: "SHADE" if you press stop button it goes to shade :)