madmartin/Jarolift_MQTT #45: Reboot command

yasars, 2021-11-11:
Hi can you tell me the reboot command ? <Ip>/system.html ???

I need it for home assistant to restatt the jaro dongle.... After 5 days jaro doesnt work and i will send an reboot all 48h ...

Regards

Scrath1, 2023-10-30:
Hi,
I haven't tested these things but I think you can piece it together based on the javascript code and html you can see when inspecting the page in your browser.

Inspecting the reboot button shows that it calls a javascript function called `writeConfig('restart')`.
This is the code of that function:
```javascript
function writeConfig(cmd){
	var elements = document.getElementById("configForm").elements;
	var params = "";
	for (var i = 0, element; element = elements[i++];) {
		if (element.type == "checkbox"){
			if (element.checked == true){
				params += element.name + "=true&"
			}else{
				params += element.name + "=false&"
			}
		}else{
			params += element.name + "=" + element.value + "&"
		}
	}
	var http = new XMLHttpRequest();
	http.onreadystatechange = function() {
		if (this.readyState == 4 && this.status == 200) {
			iqwerty.toast.Toast(this.responseText);
		}
	};
	http.open("POST", "api", true);
	http.setRequestHeader("Content-type", "application/x-www-form-urlencoded");
	http.send("cmd=" + cmd + "&" + params);

}
```

Based on the final three lines I would say that you have to sent a post request to the `/api` URL with the command formatted in the same style as can be seen in the last line in the function.

When I click the button with a breakpoint set on that final line, the params seem to contain all settings from the settings page. I am not sure if you need all of those as well to restart the dongle. That is something you will have to test yourself.

This is what I get from the params variable at the end of the function. I have censored some sensitive stuff obviously
```
"ssid=********&password=********&dhcp=false&ip_0=0&ip_1=0&ip_2=0&ip_3=0&nm_0=0&nm_1=0&nm_2=0&nm_3=0&gw_0=0&gw_1=0&gw_2=0&gw_3=0&mqtt_broker_addr_0=192&mqtt_broker_addr_1=168&mqtt_broker_addr_2=1&mqtt_broker_addr_3=103&mqtt_broker_port=1883&mqtt_broker_username=********&mqtt_broker_password=********&mqtt_broker_client_id=&mqtt_devicetopic=jarolift&master_msb=********&master_lsb=********&learn_mode=true&serial=********&set_and_generate_serial=false&devicecounter=149&set_devicecounter=false&save=save&restart=restart the dongle&"
```

eHarry87, 2023-10-30:
for me this is working in HA:

rest_command:
  jaroliftdonglereset:
    url: http://***ip to your dongle***/api
    method: POST
    content_type: "application/x-www-form-urlencoded"
    payload: "cmd=restart"

closed 2023-10-31 by yasars