madmartin/Jarolift_MQTT #49: Admin mode error (blank page)

claude-boulard, 2022-10-13:
Hi all,

I've followed the given instructions, but after reboot in admin mode, connection to the ESP access point and then connection to the webserver (http://192.168.4.1/), I get a blank page with just '404: Not Found'.
In the mean time on the serial monitor, a message is displayed : File not found: did you upload the data directory?

My HW is:
- NodeMCUJ v3,
- CC1101 tranceiver.

From this forum (https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/) the appropriate file update tool is the 'LittleFS Filesystem Uploader', this is the one I've installed and used to update the ESP file system.
So far this was apparently ok : during update it displays the list of detected files (the ones in the 'data' folder), and at the end I got a message 'LittleFS Image Uploaded'.

So right now I can't go further (configuring via admin mode). Any help/suggestion will be greatly appreciated.

Thanks.
Claude

madmartin, 2022-10-14:
Hello Claude,

I did not work for a long time on this project.
At time of programming this, the filesystem on flash was named SPIFFS which is different from what you created (LittleFS). 
I know that SPIFFS is declared as deprecated (https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#spiffs-deprecation-warning) but this project currently uses it.
So please re-create the filesystem as SPIFFS and put the files into it. This should make the webserver work.

madmartin, 2022-11-20:
@claude-boulard Did you fix your issue? Short feedback is always appreciated.