madmartin/Jarolift_MQTT #51: Feature: Time Scheduler

HerrRiebmann, 2022-11-04:
Currently I am not planning to use a home automation with MQTT.
But I want to centralize the timing of my Jarolift shutters. It´s a pain in the a**, changeing it several times a year. And on top of that, the devices will all have a little delay...
And the ESP8266 is running all the time, anyways!

So I started to enhance the code with some functionality. For now, just as a proof of concept and not very "clean coded".

[image]

What´s missing:

- [ ] Loading Setup (SPIFFS)
- [ ] Storing Setup (SPIFFS)
- [ ] Optimizing HTML & JS
- [ ] Optimizing Arduino Code

If you are interested in this as an optional feature, I can create a Pull request. Otherwise I´ll just leave my code here, when it´s ready.

HerrRiebmann, 2022-11-06:
After several hours, I´ve got it going... partially...
But first things first:

- [X] Loading Setup (SPIFFS)
- [X] Storing Setup (SPIFFS)
- [X] Optimizing HTML & JS
- [ ] Optimizing Arduino Code

The scheduler works fine! It will be executed within the first second of the scheduled minute and recognizes the time, to prevent several executions.
Changing the schedule via web interface, resets the next scheduled time, to force a new calculation.

Since I packed everything in a 3D printed case, I´ve also introduced OTA Updates.

Now the downside: On my testing, connected to the notebook, everything worked fine! Changing to a wallcharger, the website only loads partially, controlling the shutter by [Http Shortcuts](https://http-shortcuts.rmy.ch/) (_a great Android App to send webrequests_) just worked sometimes.
After changing the USB cable, It got better. But still not reliable.
Sometimes I can remote control 3 shutter in a row, sometimes, none of them responds. It´s quite hard to determine if it´s the WiFi, the distance to the shutter, antenna orientation, power or whatever.
I cannot reach the WAF* with a solution like that :(

*WomanAcceptanceFactor

HerrRiebmann, 2022-11-08:
Since OTA does not work reliable in my network, I´ve added an upload Button, which can handle regular files, which will be created/overwritten on the SPIFFS or *.bin files for firmwareupdates.
The Arduino code got better, but is still not a beauty...

Still some testing and optimization necessary (_Some shutter does not listen to the remote or this device, anymore_). Finding the perfect spot to reach every motor. Go into a long-time test. Create a pullrequest.

Until then, here´s my branch with the current state:
https://github.com/HerrRiebmann/Jarolift_MQTT

wunderbaum, 2022-11-12:
As far as I understand this enhanced version, even if you do not use mqtt the code is still in it an useable?

HerrRiebmann, 2022-11-12:
Yes, I left MQTT untouched. It should still work as before.

OTA via Upload-Button currently deletes the whole SPIFFS, which I could not figure out yet. But everything else works quite good.

HerrRiebmann, 2022-11-18:
Allright, I´ve tried to outsource some code from the `setup()` and `loop()` functionality to make it a bit more readable.
The schedules are working reliable now, independent of the number of shutter (_sometimes a shutter was skipped_).

The ranges between my shutter were too big, so I introduced a boolean to differ between two devices and their schedules.txt-files.
During the next days, I will go on testing and create a pullrequest to merge my changes.

closed 2022-11-18 by HerrRiebmann