madmartin/Jarolift_MQTT #52: Backup?

hubsi5, 2022-11-20:
Kann man die Einstellungen irgendwie sichern?
Ich würde gerne einen anderen Wemos versuchen wollen, da sich das Ganze bei mir leider öfter aufhängt.
Würde das gehen, ohne dass man die Rollläden neu anlernen muss?

madmartin, 2022-11-20:
> Kann man die Einstellungen irgendwie sichern?

Leider nein.

> ... ohne dass man die Rollläden neu anlernen muss?

Ja.
Auf dem neuen Gerät die gleiche Reihenfolge der Shutter verwenden, und dann den DeviceCounter  und "channel serial number prefix" vom alten Gerät übernehmen (Häkchen bei "save new device counter" nicht vergessen).
Seriennummer und DeviceCounter ist das, was sich die Rolläden beim "anlernen" abspeichern.
Für mqtt ist natürlich noch das "device topic" wichtig.

hubsi5, 2022-11-20:
oh super danke!
Wäre eine Version mit einem ESP32 auch denkbar?
Die laufen bei mir generell stabiler

madmartin, 2022-11-20:
Das weiß ich nicht, hab es noch nicht probiert.

hubsi5, 2022-11-20:
Ok ich kenne mich da leider zu wenig aus.
Hast du nie Probleme mit der Stabilität?