madmartin/Jarolift_MQTT #53: Nur Up wird erkannt

hubsi5, 2022-11-27:
Ich habe nun meinen zweiten Aufbau kopiert (channel serial number prefix und device counter) vom altem Aufbau übernommen.
Hat funktioniert bei 5 Rollläden.
Einer hat Probleme damit. Er erkennt nur den UP Befehl. Die anderen nicht.
Kann man das irgendwie wieder hinbiegen ohne neu anzulernen?
Problem ist, dass ich dort sehr schwer hinkomme zum Motor.

madmartin, 2022-11-27:
Hi,
so ein Problem ist mir noch nie untergekommen. Da hab ich auch keine andere Idee als neu anlernen.
Alle Geräte sind übrigens AFAIK direkt nach dem "Strom an" im Lernmodus, so daß evtl. die Sicherung raus- und reinmachen hilft. (wenn nicht mehrere Rolläden an einer Sicherung hängen....).

hubsi5, 2022-11-27:
Hängt wohl irgendwie mit dem Counter zusammen. Wenn ich den verändere klappt es dann ein paar mal, und dann wieder nicht.
Was ist eigentlich wenn man nun eine Woche nur mit der original Fb die Rollläden steuern würde?
Dann würde der Counter in diesem Programm auch zu Problemen führen, oder?

blissi, 2022-12-31:
Das würde mich auch interessieren. Und die Zeitspanne ab der das problematisch werden würde

madmartin, 2022-12-31:
Soweit ich das Keeloq Protokoll verstanden habe besteht das "anlernen" darin, sich von einem Sender dessen Seriennummer plus den Counter zu merken. Jedes gesendete Kommando enthält Seriennummer, Counter plus Kommando, das ganze verschlüsselt mit dem Masterkey.
Der Counter kann dann je Sender in einem gewissen Bereich variieren, damit die Kommandos akzeptiert werden (das ist so gedacht falls der Empfänger mal ein paar Datagramme eines Senders nicht mitbekommt). 
Ein Empfänger kann mehrere diese Paare (Seriennummer+Counter) "lernen" - also auf mehrere "Fernbedienungen" hören.
Jetzt ist die Frage wie groß das "Fenster" für die Counter ist - das weiß ich auch nicht. Vielleicht steht was in der ursprünglichen keeloq-Dokumentation, aber dann wissen wir nicht ob man sich auch daran gehalten hat.....
Eines sehe ich aber als klar an: Mit "Zeitspannen" hat das Protokoll nichts zu tun.
Ich persönlich habe die Beobachtung gemacht daß mein Projekt hier zwar 100% stabil läuft, aber nicht 100% zuverlässig, insbesondere bei rasch aufeinander folgenden Befehlen. Ich habe in die Home Assistant Automatisierungen, die die Befehle geben, immer 0,5 sek. pausen eingebaut. Damit geht es meist gut.

hubsi5, 2022-12-31:
Eventuell wäre es ja einen Versuch wert, das Ganze auf einem ESP32 zu realisieren.
Die sind stabiler und vielleicht klappt es dann zuverlässiger.
Ich kann da aber leider nicht helfen, da ich davon keine Ahnung habe