madmartin/Jarolift_MQTT #54: ESP8266 only stable for a week

AIXruth, 2022-12-25:
My ESP8266 is hanging after round about a week. I couldn't find the reason yet. 
Will a second remote controller in use negativ influence the device counter?

HerrRiebmann, 2023-01-24:
Does the ESP hung up or does the shutter stop to recognize the ESP?

I have two ESP32´s because the shutters were too far away (different story). After a while, some shutter did not work properly.
Since I know they all share one counter and listen to the signals, even if they were not controlled, I changed the "channel serial number prefix". Seperated the devices on the ESP´s and learned them new on the shutters.

And still, the shutter, which will listen to the commands from the "wrong" device, will stop working after a while. I have to re-learn it or just wait for about 10 minutes to be able to control it again.

The WAF is quite bad with a sometimes working solution like that :(
Two seperate devices, different serial number and counter. No channel overlaps.
Additional remote controller currently not in use...

AIXruth, 2023-01-24:
After a few days the ESP8266 has no control over any shutter anymore. The shutter-buttons are still working and logging, but the shutters itself doesn't react. Using the "save" or "reboot" button on the system page results in a hanging ESP. The IP doesn't respond anymore till I reset the device using the power switch. Afterwards I have to re-learn the shutters again.

HerrRiebmann, 2023-01-25:
On a classic Arduino I´d say you have an EEPROM issue. But the ESP doesn´t have one!
I´d try to grab the latest ESP Core and Libraries, to ensure it´s not a bug.
If this does not work fine, I´d get an ESP32 and give it a try. For 5-6 bucks, no big deal.

At least, my ESP32´s were running for months staight. Only one or two shutters go crazy.

johi-1, 2023-04-21:
Do I have to change anything before flashing an ESP32 with your software?

HerrRiebmann, 2023-04-23:
No, you just need to flash it onto it with the ESP right board settings.

HerrRiebmann, 2023-12-04:
@atze09 I think so, but I cannot tell you how much code you need to change.
Or if the firmware "rerouts" the WiFi traffic automatically to the ethernet. I have not worked with that version until yet.

AIXruth, 2023-12-18:
@atze09: The current code is based on a WIFI connection. If you re-program the network part, it will be possible.

Based on my first post, I have created the following workaround that works for me:
I switch on the ESP via a switch socket, switch the shutters twice at intervals and then switch the ESP off again.
And yes, a more reliable ESP would also do the job without switch socket.

atze09, 2024-01-15:
Migration to esp32 eth01 or in general to esphome project world be awesome.
My coding skills are definitely to low for according this.