madmartin/Jarolift_MQTT #57: Homeassistant MQTT Einbindung

bhn01, 2023-02-26:
Gibt es irgendwo eine Beschreibung wie die Einstellungen in HA zu erfolgen haben ?
Ich habe bisher den MQTT Broker nebst User und Passwort am laufen, aber der Dongle wird in der Integration nicht erkannt. Muss man die Entitäten manuell anlegen ? Für Erhellung wäre ich dankbar, ich suche schon seit Stunden rum.
Der Dongle ist aber soweit am laufen.

stalakerob, 2023-02-26:
Hi, das geht nicht automatisch. Ich habe das in configuration.yaml so konfiguriert:
`mqtt:
  cover:
    - name: "Jarolift"
      command_topic: "cmd/jarolift/shutter/0"
      availability_topic: "tele/jarolift/LWT"
      state_topic: "stat/jarolift/shutter/0"
      value_template: >
        {% if value == "100" %}
          open
        {% else %}
          closed
        {% endif %}    
  #    position_topic: "stat/jarolift/shutter/0"
  #    position_open: 0
  #    position_closed: 100
      payload_open: "UP"
      payload_close: "DOWN"
      payload_stop: "STOP"
      payload_available: "Online"
      payload_not_available: "Offline"
`
HA erzeugt eine Entität names cover.jarolift

bhn01, 2023-02-26:
vielen Dank! 
Die Entität ist da. Wie lässt sich denn jetzt die Automatisierung umsetzen ? Ich bin leider gerade am Anfang mit MQTT...

stalakerob, 2023-02-26:
MQTT handelt eigentlich HA für dich. Durch die Konfiguration kennt es jetzt die MQTT Details um das Cover steuern zu können. Du kannst das cover einfach als Entität in das GUI einfügen und es dann manuell steuern oder über HA Automations. Die services findest du hier:  [https://www.home-assistant.io/integrations/cover/](url)
Alles über Automations hier: [https://www.home-assistant.io/docs/automation/](url)
Die HA Community hilft dir bei allen Fragen: [https://community.home-assistant.io/](url)

johi-1, 2023-03-02:
Hallo,
habe gleiches oder hnliches Problem. Dongle läuft. Rolladen fährt, aber bekomme ihn nicht in Homeassistant. Habe versucht den text in config. yaml einzugeben. Bekomme aber Fehlermeldungen für die Zeile value template:  >
Kann mir jemand helfen?
Gruß 
HH

bhn01, 2023-03-02:
mqtt:
 cover:
  - name: "Jarolift" 
    command_topic: "cmd/jarolift/shutter/0" 
    availability_topic: "tele/jarolift/LWT" 
    state_topic: "stat/jarolift/shutter/0" 
    value_template: > 
      {% if value == "100" %} open
      {% else %} closed {% endif %}     
    payload_open: "UP" 
    payload_close: "DOWN" 
    payload_stop: "STOP" 
    payload_available: "Online" 
    payload_not_available: "Offline"

stalakerob, 2023-03-02:
Es kann an der Formatierung liegen. Yaml ist sehr sensitiv bzgl Einrückungen etc. Leider funktioniert hier die Code Formatierung irgendwie nicht richtig. Es sollte so aussehen:
[image]

bhn01, 2023-03-02:
Hallo Johi,
nimm mal statt dem Fileeditor den Studio Code Server. Der zeigt dir auch die Formatierungsfehler an.

johi-1, 2023-03-03:
Vielen Dank,

hat funktioniert. Rolladen fährt jetzt über HA.

Vile Grüsse
jh

Von: bhn01
Gesendet: Donnerstag, 2. März 2023 19:40
An: madmartin/Jarolift_MQTT
Cc: johi-1; Comment
Betreff: Re: [madmartin/Jarolift_MQTT] Homeassistant MQTT Einbindung (Issue#57)

Hallo Johi,
nimm mal statt dem Fileeditor den Studio Code Server. Der zeigt dir auch die Formatierungsfehler an.

johi-1, 2023-03-03:
Das wars
Danke

Von: Robert
Gesendet: Donnerstag, 2. März 2023 19:39
An: madmartin/Jarolift_MQTT
Cc: johi-1; Comment
Betreff: Re: [madmartin/Jarolift_MQTT] Homeassistant MQTT Einbindung (Issue#57)

Es kann an der Formatierung liegen. Yaml ist sehr sensitiv bzgl Einrückungen etc. Leider funktioniert hier die Code Formatierung irgendwie nicht richtig. Es sollte so aussehen:

Stevelande, 2023-06-24:
Habe hier mal eine Beispiel yaml. Mit Statusanzeige für die Rollos. Vielleicht kann es ja jemand gebrauchen.
`mqtt:
  cover:
    - name: "Rollo 1"
      command_topic: "cmd/jarolift/shutter/0"
      availability_topic: "tele/jarolift/LWT"
      payload_open: "UP"
      payload_close: "DOWN"
      payload_stop: "STOP"      
      payload_available: "Online"
      payload_not_available: "Offline"

    - name: "Rollo 2"
      command_topic: "cmd/jarolift/shutter/1"
      availability_topic: "tele/jarolift/LWT"
      payload_open: "UP"
      payload_close: "DOWN"
      payload_stop: "STOP"
      payload_available: "Online"
      payload_not_available: "Offline"

    - name: "Rollo Küche"
      command_topic: "cmd/jarolift/shutter/2"
      availability_topic: "tele/jarolift/LWT"
      payload_open: "UP"
      payload_close: "DOWN"
      payload_stop: "STOP"
      payload_available: "Online"
      payload_not_available: "Offline"

    - name: "Rollo Schlafzimmer"
      command_topic: "cmd/jarolift/shutter/3"
      availability_topic: "tele/jarolift/LWT"
      payload_open: "UP"
      payload_close: "DOWN"
      payload_stop: "STOP"
      payload_available: "Online"
      payload_not_available: "Offline"
  
  button :
    - unique_id: R1_shade_btn
      name: "Rollo 1 Schatten"
      command_topic: "cmd/jarolift/shutter/0"
      payload_press: "SHADE"
      
    - unique_id: R2_shade_btn
      name: "Rollo 2 Schatten"
      command_topic: "cmd/jarolift/shutter/1"
      payload_press: "SHADE"
      
    - unique_id: Rk_shade_btn
      name: "Rollo Küche Schatten"
      command_topic: "cmd/jarolift/shutter/2"
      payload_press: "SHADE"

    - unique_id: Rs_shade_btn
      name: "Rollo Schlafzimmer Schatten"
      command_topic: "cmd/jarolift/shutter/3"
      payload_press: "SHADE"

  sensor:
    - name: "Rollo 1 Zustand"
      state_topic: "stat/jarolift/shutter/0"
      value_template: >-
        {% if value == "0" %}
          offen
        {% elif value == "100" %}
          geschlossen
        {% elif value == "90" %}
          Schatten
        {% else %}
          unbekannt
        {% endif %}
    
    - name: "Rollo 2 Zustand"
      state_topic: "stat/jarolift/shutter/1"
      value_template: >-
        {% if value == "0" %}
          offen
        {% elif value == "100" %}
          geschlossen
        {% elif value == "90" %}
          Schatten
        {% else %}
          unbekannt
        {% endif %}
    
    - name: "Rollo Küche Zustand"
      state_topic: "stat/jarolift/shutter/2"
      value_template: >-
        {% if value == "0" %}
          offen
        {% elif value == "100" %}
          geschlossen
        {% elif value == "90" %}
          Schatten
        {% else %}
          unbekannt
        {% endif %}
    
    - name: "Rollo Schlafzimmer Zustand"
      state_topic: "stat/jarolift/shutter/3"
      value_template: >-
        {% if value == "0" %}
          offen
        {% elif value == "100" %}
          geschlossen
        {% elif value == "90" %}
          Schatten
        {% else %}
          unbekannt
        {% endif %}
`