madmartin/Jarolift_MQTT #62: Jarolift_MQTT vergisst Rolladenmotoren

branka5, 2023-05-10:
Hallo,

Ich habe das Problem, dass Jarolift _ MQTT immer wieder einen Rollladenmotor vergisst und ich diesen dann nur noch über die Jarolift Fernbedienung steuern kann. Das Aus- und Einschalten des Jarolift_MQTT als auch das Stromlos machen der Rolladenmotoren führt ebenso wenig zum Erfolg wie der Versuch des Neueinlernens. was kann ich tun?

Stevelande, 2023-07-05:
Hast du [Diese](https://github.com/madmartin/Jarolift_MQTT/compare/master...HerrRiebmann:Jarolift_MQTT:master) Version installiert?

Stevelande, 2023-07-12:
Habe alles nochmal neu verlötet und diese Version drauf, seit ca. 3 Wochen läuft alles ohne Ausfälle.

blissi, 2023-07-12:
Danke, dann probiere ich das auch einmal aus.