madmartin/Jarolift_MQTT #63: Fertigmodul

nuiler, 2023-06-28:
Hat jemand eventuell ein Fertigmodul oder Bausatz zu kaufen?

In diesem Forum kann ich mich nicht registrieren und den Verkäufer anschreiben.
[Link][http://www.bastelbudenbuben.de/forum/topic/platine-fuer-wemos-d1-und-cc1101-und-gehaeuse/?part=5](url)

oder hat jemand von ihm Kontaktdaten.

Stevelande, 2023-07-05:
Ich glaube der Tread ist schon lange tot. Habe es selbst gebaut Wemos D1 Mini und blaues cc1101 Modul

Stevelande, 2023-07-05:
Gib mal ne Mail Adresse. Öffentlich Posten will ich nicht.

nuiler, 2023-07-05:
> Von "Stevelande" ***@***.***>
> An "madmartin/Jarolift_MQTT" ***@***.***>
> Cc "nuiler" ***@***.***>; "Author"
> Datum 05.07.2023 14:24:11
> Betreff Re: [madmartin/Jarolift_MQTT] Fertigmodul (Issue #63)

***@***.***

Subbilein, 2023-07-19:
Hallo Zusammen,
Ich hätte auch Interesse an den Infos, MasterMSB und evtl. weitere Infos zum Selbstbau mit Wemos D1 Mini und CC1101 Modul.
Es wäre doch gelacht, wenn ich nach 4 Jahren nicht endlich auch meine Cover in HA rein bekomme :D

Stevelande, 2023-07-19:
Bin gern behilflich!

Subbilein, 2023-07-19:
Dankeschön.
Du kannst mir gerne auf mr.subba@web.de Infos zusenden.