madmartin/Jarolift_MQTT #64: Öeoterplattine

Murdock79, 2023-07-10:


Subbilein, 2023-07-19:
Ich hätte auch Interesse :)
Meine Löt-Skills sind auch begrenzt.

Murdock79, 2023-07-19:
Habe mir das über eine Software konstruiert und in China anfertigen lassen. Habe gleich 5 Platinen gefertigt bekommen. Falls Interesse besteht und die Platinen funktionieren, kannst du gerne eine Platine von mir haben. Muss noch warten, bis sie zu mir geschickt werden.

Subbilein, 2023-07-19:
Das wäre super.
Kannst du mich dann auch mit dem MasterMSB Key supporten ?
Danke schonmal!

Murdock79, 2023-07-19:
Ja, kein Problem. Ich helfe dir. Habe sehr lange gebraucht, um das zu checken und gebe mein Wissen gerne weiter. Schreib mir einfach privat.

Subbilein, 2023-07-19:
Vielen Dank. Wie kann ich dir privat schreiben?Am 19.07.23 um 16:42 schrieb Murdock79
                
            
            
                
                    Von: "Murdock79" ***@***.***>Datum: 19. Juli 2023An: "madmartin/Jarolift_MQTT" ***@***.***>Cc: "Subbilein" ***@***.***>,"Comment" ***@***.***>Betreff: Re: [madmartin/Jarolift_MQTT] Leiterplatine (Issue #64)
                
                
Ja, kein Problem. Ich helfe dir. Habe sehr lange gebraucht, um das zu checken und gebe mein Wissen gerne weiter. Schreib mir einfach privat.

Murdock79, 2023-07-19:
> Vielen Dank. Wie kann ich dir privat schreiben?

Email sollte nun in meinem github Profil angezeigt werden.

Subbilein, 2023-07-19:
Wird leider nicht angezeigt. 
dann bitte an mr.subba@web.de

Stevelande, 2023-07-19:
Habe schon eine laufende Version, dennoch wäre ich Interessiert an der Platine. Hast du etwas mehr Infos? Schaltplan, Komponenten?
Mein Wissen gebe ich auch gerne weiter.

Murdock79, 2023-07-20:
Bin grad im Urlaub, und werde den Schaltplan hier posten, wenn es funktioniert.
Meine Mail Eintraum79@googlemail.com
Hat jemand die Software mit Platformio installiert und kann mir dabei helfen?