madmartin/Jarolift_MQTT #70: Question to Jarolift_MQTT project

elu, 2024-06-24:
Hi madmartin,

i am trying since weeks to find the Jarolift masterkey without any success. I hope you can help me. I tried already some keys, but no reaction from the learn-button on the web-client even if i pressed a second before the button on the TDEF-device.
You can send me an email to elu2@post.com or leave me a PN on my github account.

I would really appreciate your help.

br
elu

t4e-dev, 2024-07-02:
https://www.google.com/search?q=jarolift+msb+pastebin

elu, 2024-07-07:
Hallo Robin,

Danke für den Tip. Werde diesen Masterkey gleich probieren, wenn ich die hoffentlich funktionierenden CC1101-Module erhalte.

Mit freundlichen Grüßen
Erwin

https://www.google.com/search?q=jarolift+msb+pastebin

gremei, 2024-07-07:
Hi elu,

my first attempt was also this project, but since it was not stable enough, I was searching for an alternative solution.
If you are running a Homeassistant and own a everhome box, you can integrate it according following discord channel solution/integration: https://discord.com/channels/1179646674692485232/1207540418800132126

JanLoebel, 2024-07-29:
@gremei I can not access that channel, can you post some more information or the discord server itself?