madmartin/Jarolift_MQTT #72: Jarolift Dongle - 404: Not Found

Muskelbiber1, 2024-11-16:
Hi, I completed the setup and am now at the point of starting the dongle and connecting via Wi-Fi.
I followed the manual step by step, and the LED is blinking.

I can see and connect to the dongle, but when I try to reach 192.168.4.1, the browser shows me a 404: Not Found error.

The only thing I did differently compared to the manual is the step regarding "SPIFFS":

When I change "Flash size," the only option available is: "4MB (FS:1MB OTA: ~1019KB)" instead of "4M (1M SPIFFS)."
After flashing, when I try to use "ESP8266 Sketch Data Upload," I get the error: "SPIFFS Error: esptool not found."

I’ve already reinstalled everything twice with different Arduino versions (1.x and 2.x), but the issue persists.

There are any ideas?

Biber

Blastmun, 2024-11-17:
I encounter another problem after this step but in fact your problem comes from the fact that upload SPIFFS does not work.

Muskelbiber1, 2024-11-17:
Awesome, it works! Thanks!
Now I found another problem, My SSID owns a "&" like "A&B"
When I what to save my SSID it only save the "A" and the rest is gone.
Do you have an idea how to fix that?

I already found issue #65 but I dont understand correctly what to do :/

dewenni, 2025-01-13:
@Muskelbiber1 if you like, you can also try my project, that can be seen as a successor of this cool project here.
https://github.com/dewenni/ESP32-Jarolift-Controller